leonardt/hwtypes #16: Make bitvector ABC

cdonovick, 2019-01-17:
We should create a BitVector ABC to aid abstraction of the BV types.

This ideally would provide mixin methods where appropriate, e.g. define  `bvnot` and `bvand` get `bvnand` for free.  

Crazy idea: I think we could get away with requiring only `__getitem__`, `concat`, and `bvnand`  for all of the `bv*` methods but that would lead to inefficient code so we probably don't want to encourage that.

We might want to add a make_constant method, this would make abstracting over Bit Vector types easier when there constructors are not gaurenteed to be uniform (e.g. a smt_bitvector needs to bound to a solver https://github.com/phanrahan/peak/blob/mapper/peak/mapper/SMT_bit_vector.py)

```Python
def make_constant(self, value:int, num_bits:typing.Optional[int] = None) -> type(self):
  if num_bits is None:
    num_bits = self.num_bits
  #else:
  #  if a type with a bound size (e.g. Bit) raise an exception 
  return type(self)(value, num_bits)
```

phanrahan, 2019-01-17:
These all seem like good ideas.

@cdonovick, any chance you could draft an abc class for bitvector so we can make all this more concrete?

cdonovick, 2019-01-17:
@phanrahan okay will draft one

leonardt, 2019-02-28:
Closed by https://github.com/leonardt/bit_vector/pull/17, let's open more specific issues as they arise.

closed 2019-02-28 by leonardt