leonardt/hwtypes #31: bitvector.random

phanrahan, 2019-05-18:
if I do Bits16 = BitVector[16]

I can't do Bits16.random() to get a random number. I seem to need to call BitVector.random(16). 

Am I missing something? I am contantly generating number numbers for tests, so this would be a nice feature to have.

I am thinking it would also be good to create options for different distributions of random numbers. For example, SIntVector.random(width, sampler="stratified") which ensures that half the random numbers are positive and half are negative. Similarly for FPVector.

cdonovick, 2019-05-20:
@phanrahan I agree it would make sense to make the random method a classmethod instead of a staticmethod (already the case FPVector).

I am not quite sure what it means to ensure half the random numbers are positive and half are negative on a method that returns a single number.

phanrahan, 2019-05-20:
We could alternate generating positive and negative numbers. Just make sure for a small number of calls that we stratify the samples. Mostly, I want to create a framework for using better sampling methods.

cdonovick, 2019-05-20:
@phanrahan I created an SMT based constrained random generator for fault.  However, the constraints are on individual samples not over a set of samples. 
https://github.com/leonardt/fault/blob/master/fault/random.py
https://github.com/leonardt/fault/blob/master/tests/test_constrained_random.py
For "stateful" constraints it make sense to change the api from
```
bv = BitVector.random(size)
```
to
```
for bv in BitVector.random_generator(size, sampler=sampler, num_samples=N):
   ...
```
or
```
for bv in sampler(size, N):
   ...
```
tagging @leonardt