leonardt/hwtypes #37: Possibly factor out FP into its own package

makaimann, 2019-05-21:
It would be not to have the extra dependencies on `libmpfr-dev` and `libmpc-dev` in CoSA when we will likely not being using the floating point features very often (if at all).

For now, it's not a problem, but it would be nice if the FP stuff was separate, assuming that's not too difficult for you all.

cdonovick, 2019-05-21:
Similarly I believe the SMT bitvectors should also be factored so non-smt consumers don't require SMT

currently I imagine the package structure as:
hwtypes: python dependencies only
   bit_vector*
   adt*
   fp_vector_abc.py

hwtypes[smt]: z3/pysmt dependencies
   z3*
   smt*

hwtypes[float]: mpfr dependencies.
   fp_vector.py

leonardt, 2019-05-28:
@cdonovick this is now blocking @ankita0805 because the TSMC VMs (where she's running PD tests) are running an old version of RHEL (6) that only provide MPC 0.8 and MPFR 2.4.1. `gmpy2` requires MPFR 3.1.0 and MPC 1.0.0.  We're investigating a workaround to install a newer version of those libraries, but relaxing this dependency by making it optional would help

leonardt, 2019-05-29:
For now we unblocked ankita by running a hwtypes branch with the fp dependency removed because I think she doesn't need those features, we can lower the priority for this issue unless something else comes up.