leonardt/hwtypes #49: [feature-request] Array Type

cdonovick, 2019-06-07:
Add an array type to better match the magma type hierarchy.

phanrahan, 2019-10-05:
Could we bump this in priority? This would be very useful for the design space exploration paper.

cdonovick, 2019-10-05:
I can bang out a provisional versional.  What operations should the array support?  Just getitem?

phanrahan, 2019-10-05:
getitem for sure.

We should also be able to declare an array as an argument to a peak method. And construct an array from a set of values.

Thanks!

cdonovick, 2019-10-05:
Would a Tuple work for your purposes?
It will be a bit verbose but it works now.

phanrahan, 2019-10-05:
I can make a tuple work. But an array would be a lot cleaner. The problem is I am trying to write a generator, and tuples might require macros. ;)

cdonovick, 2019-10-05:
I don't think they should.  Array[N, T] is Tuple[T for _ in range(N)].

cdonovick, 2020-07-23:
related to https://github.com/phanrahan/magma/issues/752