leonardt/hwtypes #56: [feature request] dynamic constructor for adt.Product

rdaly525, 2019-07-05:
`new_product(class_name : str, field_dict : tp.Mapping[str,type])`

rdaly525, 2019-07-08:
My current hack for this
```
def _new_product(class_name,field_dict):

    class_str = f"class {class_name}(Product):"
    for name,t in field_dict.items():
        class_str += f"\n    {name}={name}"
    class_str += "\n"
    exec_globals={**dict(field_dict),"Product":Product}
    exec_locals = {}
    exec(class_str,exec_globals,exec_locals)
    return exec_locals[class_name]
```

cdonovick, 2019-07-10:
I totally forgot we already have this feature.

```
Product('P', field_name=field_type)
class P(Product):
  field_name = field_type
```

Are equivalent

cdonovick, 2019-07-10:
However, I am removing the call version in favor of a method based version

PR #64 (2019-07-11): Dynamic product
Resolves #56
closed 2019-07-11 by cdonovick