leonardt/hwtypes #57: getting the 'nth' field from an adt.Product

rdaly525, 2019-07-06:
It would be great if field_dict and value_dict in adt.Product were an OrderedDict. Alternatively, keeping a ordered list of field names would be sufficient.

@cdonovick, is there any ways to get the 'nth' field from a Product currently?

cdonovick, 2019-07-08:
@rdaly525 all dicts are ordered in 3.7 and have been in CPython since 3.6. The nth fieldname is the nth key.

https://docs.python.org/3/library/stdtypes.html#typesmapping
"Dictionaries preserve insertion order. Note that updating a key does not affect the order. Keys added after deletion are inserted at the end. ... Changed in version 3.7: Dictionary order is guaranteed to be insertion order. This behavior was an implementation detail of CPython from 3.6"

Also if you just want the nth field you can tuple index products.

cdonovick, 2019-07-08:
As the order should matter in comparison of field dicts I will switch to an OrderedDict

closed 2019-07-10 by cdonovick