leonardt/hwtypes #71: Possible ADT naming change for more consistency

rdaly525, 2019-07-17:
I think we should rename our adt types to be the following:

(sum types)
Variant
TaggedUnion 

(product types)
Tuple
Record

Currently, Sum behaves like Variant, Product behaves like Record, and Tuple behaves like Tuple.

This would mean that there is no class called Sum or Product for constructors.

Thoughts on this? It is a bit weird to currently have 'Sum' refer to the unnamed version of an ADT Sum type whereas 'Product' refer to the named version of an ADT Product type.

rdaly525, 2019-07-24:
@leonardt @cdonovick @phanrahan 

This is the issue I mentioned about potentially renaming ADT classes.

phanrahan, 2019-07-30:
I like the idea of unifying the system so all variants of tagged vs non-tagged and product vs sum are possible.

Can we add class names to Sum? And support both indexed and named access?

cdonovick, 2019-07-30:
@phanrahan 
We currently support named access
```
Sum[A, B].A == A
```
But we don't have syntax for having specified names e.g.:
```
class Foo(TaggedUnion):
   bar = A
   foobar = B
Foo.bar == A
```

phanrahan, 2019-07-31:
I thought about this some more. I personally don't think we need a tagged variant of Sum types. There is no need to access an instance of a sum type except via match. Product types are different in that we have to access the fields of an instance.