leonardt/hwtypes #100: dunderscore operators not defined in base class

leonardt, 2019-10-21:
Right now the dunderscore operators (e.g. __add__) are not defined in the ABC, instead they are defined as `bvadd`, etc.., then the concrete classes implement them but also have to define the dunderscore methods (e.g. https://github.com/leonardt/hwtypes/blob/master/hwtypes/z3_bit_vector.py#L471-L491 and https://github.com/leonardt/hwtypes/blob/a184707062497a1b68e1af2c293c744482833d6f/hwtypes/bit_vector.py#L341-L473)

It seems that we could have these defined in the abstract base class to refer to the abstract methods. The reason it seems for them not being there is that the BV class has a different implementation than the z3 class (special handling for an exception case). Should we instead just have the BV class override the dunderscore methods but have a default version that dispatches to `bv<name>` methods? Basically, asking if we can have https://github.com/leonardt/hwtypes/blob/master/hwtypes/z3_bit_vector.py#L471-L491 be the default implementation inherited from the ABC.

leonardt, 2019-10-21:
I think I see the issue now, basically, we want to return notimplemented in the case that an error occurs to try and dispatch to "other"'s implementation to see if it can handle the inconsistent size. To do this, we need to define the operators in the context of the concrete implementation.

phanrahan, 2019-10-21:
When I first implemented the `BitVector` class, I added `bv*` methods that were intended to exactly reproduce the SMTBitVector semantics.

cdonovick, 2019-10-22:
@leonardt We can add the dunder methods to the base classes.  The way things are set up in bitvector is not totally by design. 

To get the AssembledADT to work I needed bitvector to return NotImplemented.  I couldn't just change my coerce function because it used in a bunch of places.   Further I didn't want the bvmethods to return NotImplemented.  

In any event I think this will be addressed by the next family construction, although we will need to agree on a set of operators for AbstractBitVector.