leonardt/hwtypes #99: kwargs interface for constructing product types

leonardt, 2019-10-21:
This is an idiom in a lot of exiting magma code, not sure if we want to support it (note this is using magma's Tuple, which is changing to Product for named fields in the hwtypes branch):
```
m.Tuple(x=m.In(m.Bit), y=m.Out(m.Bit)
```
This allows one line definition of types rather than having to explicitly define a new subclass of `Product`, e.g.
```
class T(m.Product):
    x = m.In(m.Bit)
    y = m.Out(m.Bit)
```

There is a way to do this using `type`, with
```python
T = type("T", (m.Product,), {"x": m.In(m.Bit), "y", m.Out(m.Bit)})
```

But it's not quite as convenient. One issue is that we can't use the square brace syntax since that doesn't support key/value pairs. Some options are
* `Product[dict(x=m.In(m.Bit, y=m.Out(m.Bit))]` -- use a dict as the key, but it's not hashable so we really have to use an ordereddict or something
* `Product(x=m.In(m.Bit), y=m.Out(m.Bit))` -- use a constructor interface, but this breaks the `[]` for parametrization pattern and overloads the normal constructor syntax for creating values
* `product(x=m.In(m.Bit), y=m.Out(m.Bit))` -- use a lowercase function interface, but this breaks the lowercase pattern for constructing/converting Python values (e.g. `int(x)`).

I'm not really sure if there's a good syntax choice for this feature, and at the end of the day it's a minor inconvenience to have to write out the class syntax or use the `type` constructor interface, but perhaps someone has another idea or an argument why one of the above syntaxes could work.

leonardt, 2019-10-21:
Another option: `make_Product(x=m.In(m.Bit), y=m.Out(m.Bit)`, this is probably the most explicit naming convention as a "type constructor"

rsetaluri, 2019-10-21:
I think `make_product` would be the best. It is explicit and pretty pythonic (e.g. see https://docs.python.org/3/library/dataclasses.html?highlight=dataclass#dataclasses.make_dataclass)

cdonovick, 2019-10-22:
We already have this interface
https://github.com/leonardt/hwtypes/blob/master/tests/test_adt.py#L142

closed 2019-10-22 by cdonovick