leonardt/hwtypes #101: similar software

Nic30, 2019-11-08:
Hello,

because I was not able to find project like your's I have written my [own](https://github.com/Nic30/pyMathBitPrecise).
I added reference on your project to similar project in readme section of my project, do you know about any other similar project?

It would be best if we can somehow merge our projects, just because code duplication is wrong. However I did not look in to details.

cdonovick, 2019-11-15:
@Nic30 Thanks for link!  There are a number of python hdls (PyMTL, MyHDL) I am not really familiar with them enough to tell you if they have their own python simulator (and therefore bitvector library).  

There is https://engineering.purdue.edu/kak/dist/BitVector-3.4.9.html but it is not specifically aimed at hardware.

I see you support 3-value logic, we have meant to implement this but it has not been a high priority for us. Looking through your code it doesn't look like it would be too hard to migrate to our system if it's something you would like to contribute. 

I am curious if you discover our project before you wrote your own?  If you did, did you choose to write you own because we didn't have x logic or were there other features missing or limitations that were problematic?

Nic30, 2019-11-15:
> PyMTL, MyHDL
I know for me the problem is that it is everything packed together I am developing similar tool.
There was a discussion if we can split this project so we can share the code. There was positive response but as far as I know I am the only one who is actually doing it.

> I see you support 3-value logic
Actually 3-value logic can be emulated. I do not thing it is necessary, but it is definitely nice to have.

> I am curious if you discover our project before you wrote your own?
No, I was searching for something  like this and found nothing suitable.
Several weeks after I wrote the code the github recommended me your repo...

> your code it doesn't look like it would be too hard to migrate to our system
I would gladly contribute, however it would be good to avoid C dependencies as they usually can not be installed on public servers like readthedocs.

cdonovick, 2019-11-15:
@Nic30 
> No, I was searching for something like this and found nothing suitable.
Several weeks after I wrote the code the github recommended me your repo...

Isn't that always the way it goes?

> I would gladly contribute, however it would be good to avoid C dependencies as they usually can not be installed on public servers like readthedocs.

The C dependencies have actually been an issue for us as well.  We actually wanted to make them optional but it never happened because of ... well I don't really have a good reason.  They are their for our floating point implementation but are not necessary for the core bitvector functionality. I will get that change made.

Nic30, 2019-11-16:
With C libraries it is a problem, because of free servers as I wrote.
I am trying to solve similar issue in https://github.com/Nic30/hdlConvertor/issues/51 (The C++/Python library has some Python code which needs to be shared with other projects. But we do not want to add dependency on C libraries due problems mentioned.)

For that library it seems to me that it is good solution to split it on Python only library + library with C extensions. However I am not sure how to do this in this library. Maybe use python math if C extensions are not available? But it seems to me that this would require lot of effort for very few.