leonardt/hwtypes #105: tests failing

phanrahan, 2019-12-30:
Not sure what I need to do to fix this. `bit_vector.py` doesn't exist.

```
Traceback:
tests/test_adt.py:4: in <module>
    from hwtypes.modifiers import new
hwtypes/__init__.py:7: in <module>
    from .fp_vector import *
hwtypes/fp_vector.py:7: in <module>
    from .bit_vector import Bit, BitVector, SIntVector
E   ModuleNotFoundError: No module named 'hwtypes.bit_vector'
```

rdaly525, 2019-12-30:
That is a strange error... Are you sure you are running from hwtypes:master?
Can you check if hwtypes/hwtypes/bit_vector.py is in your local repo?
Also have you tried a fresh environment?

phanrahan, 2019-12-30:
I cloned it from scratch. Now it works. Not sure what happened.

closed 2019-12-30 by phanrahan