leonardt/hwtypes #107: Too eager of casting in SMTBit.ite

rdaly525, 2020-01-08:
I would expect the following to raise a type error. Is this expected behavior or a bug?

```
p = SMTBit()
t = SMTBit()
f = SMTBitVector[1]()
#This does not raise a type error
res = p.ite(t,f)
```

cdonovick, 2020-01-08:
This basically is question about whether the following should work: `bit.ite(bv, 5)`
All of the other operators / methods currently will try to coerce there operands.  The fact the operand is a bit should not matter to this rule.

cdonovick, 2020-01-08:
I am fine with removing coercion from the ITE.  I am not okay with special casing bit.

cdonovick, 2020-01-08:
Actually I think I can block this specific behavior by first trying to cast to bit, before tying to cast to bv.

commit e6a13501 (2020-01-08): Ite cast to try to cls before BV_T

resolves #107
closed 2020-01-08 by cdonovick