leonardt/hwtypes #121: [Feature Request] Add a Union ADT type

rdaly525, 2020-01-28:
A Union ADT type is similar to a Sum type, except it would not have any tags. There would be no way to determine if a value was of a particular sub-type.

This is a good way to solve @phanrahan  icache initialization issue while still being well typed.
`icacheDataType = Union[Instr,BitVector[32]]`

I also have some auto-mapper usecases for it.

rdaly525, 2020-01-28:
Chatting with Caleb more, this will not work directly as its own independent type since one cannot cast arbitrary python types to other types. It might be able to work as a particular Assembly scheme for Sum type in which it always returns Bit(True) for match.

cdonovick, 2020-01-28:
Until you present a compelling case that works with Sum types I am not going to consider building this.

closed 2020-01-28 by cdonovick