leonardt/hwtypes #128: [RFC] BV Protocol

cdonovick, 2020-06-08:
In order to support `bit.ite(adt1, adt2)` where `type(adt1) is type(adt2)` in PEak  I think I am going to create a `_to_bitvector_`, `_from_bitvector_` protocol in the spirit of the magma protocol.   Do any of you have any feature requests or requirements that I should keep in mind while building this?

My current plan is to subset the magma protocol (modulo a remaining)
```Python3
#_to_magma_
_to_bitvector_t_(cls) -> type(BitVectorT)
#_from_magma_value_
_from_bitvector_(cls, bv: BitVectorT) -> cls
#_get_magma_value_
_to_bitvector_(self) -> BitVectorT
```

I will probably follow magmas design and make protocols inheritable classes.  However, as there will be no mixin methods actual inheritance will not be necessary and I will create an interface for registering sub classes to avoid metaclass contention. 

As of right now I would only call these methods in `ite` and not in bv constructors so that: 
```Python3
BitVector[n](val) -> TypeError
``` 
when 
```Python3
not isinstance(val, BitVector[n]) and isinstance(val._to_bitvector_(), BitVector[n])
```

I will only allow `ite` over protocol values with explicit type matches e.g given `val` from above.:
```Python3
bv = BitVector[n](...)
bit.ite(val, bv) -> TypeError
val2 = ... 
assert type(val2) is not type(val)
assert val2._to_bitvector_t_() is val._to_bitvector_t_()
bit.ite(val, val2) -> TypeError
```
Further ite will cast the value back :
```Haskell
old_ite :: (AbstractBit a, AbstractBitVector b) => a -> b -> b -> b}

to_bitvector :: (Protocol a, AbstractBitVector b) => a -> b

from_bitvector  :: (Protocol a, AbstractBitVector b) => b -> a

new_ite ::  (AbstractBit a, Protocol b) => a -> b -> b -> b
new_ite cond t f = from_bitvector $ old_ite cond (to_bitvecor t) (to_bitvector f)
```

leonardt, 2020-06-09:
Seems reasonable, if I understand the explicit type matches, it basically won't implicitly coerce two protocols to bit vector even if that would work to make the type match (basically, the user has to do the coercion explicitly themselves before calling its), so this is only convenient for when the types of the arguments to ite match (then the function does the conversion internally and also converts the result back to the original type).

leonardt, 2020-06-09:
One wrinkle: is there a reason to not support `isinstance(adt2, adt1)` instead of explicit type equality with `is`? This would make the type system more extensible, but it probably will make the type coercion rules and implementation a big headache, so I could see that as a valid reason to avoid it.

rdaly525, 2020-06-09:
What will be the relationship between the AssembledADT class and this new BitVector protocol mix-in?

cdonovick, 2020-06-09:
@leonardt the reason for the `is` is to completely avoid any coercion (even to a base type).  I ensure 
```Python3
if cond:
    x = Unsigned()
else:
    x = Signed()
x.method()
```
Has the semantics in pure python and smt by using the `PolyVector`.  Now the same approach could probably be used for protocols but it would add a lot of requirements to building a protocol type.

cdonovick, 2020-06-09:
@rdaly525 `assert issubclass(AssembledADT[adt_t, ...], BitVectorProtocol)`

closed 2020-06-25 by cdonovick